ut-ras/robomaster #33: Review .gitignore and purge project of old, no longer tracked files

calebchalmers, 2022-09-17:
Our .gitignore contains an error where all `ut-robomaster/build` files are being tracked due to a missing slash after the folder name.

Also, the `.vscode` directory should not be ignored, as this contains useful project tasks, build targets, configuration, and extensions. Some files currently in this folder should either be moved to another location or ignored, as they are machine-specific.

Once all changes are made, we need to clear the cache and re-add all files to cleanup the project.
This can be done via `git rm -r --cached . && git add .`, then committing those changes (there'll be a ton of removed build files).

doprz, 2022-09-17:
We should take care of this issue before issue #32 "Rebuild project on Linux to paths using wrong slashes", so that when we rebuild the out copy of taproot and modm with `lbuild build`, `.gitignore` will apply to the updated repo build.

doprz, 2022-09-18:
Closes #34

closed 2022-09-18 by doprz